fastmachinelearning/hls4ml #1128: User-exposed optimiser skipping

bo3z, 2024-11-14:
### Motivation
Users may want to skip certain optimisers when converting the model; the current approach of 'SkipOptimizers' doesn't work with the introduction of flows.

## Details
For more details, see: https://github.com/fastmachinelearning/hls4ml/pull/881#discussion_r1841585263

### Parts of hls4ml being affected
All backends, but the solution should be (somewhat) unified.

spacenets7, 2026-11-02:
Is anyone still actively working on this? I saw the earlier interest and closed PR #1501. The current flow execution appears to use flow.optimizers directly, bypassing the configured selection. I’d like to investigate model-local filtering and add regression coverage for dependent flows and isolation between models. Before implementing it, I’d appreciate guidance on how selection should interact with essential backend and writer passes.

spacenets7, 2026-11-02:
I reproduced the ignored SkipOptimizers setting on main at 863a4e9 using an automated local test harness.

A skipped pass still executes both directly and through a required sub-flow. In a real Vivado conversion, setting SkipOptimizers = ['vivado:optimize_pointwise_conv'] still converts a kernel-size-1 Conv1D into PointwiseConv1D.

The three skip expectations fail, while the two controls and all 28 existing flow tests pass. No fix has been applied.

Before changing execution, could you clarify how skipping should interact with essential backend and writer passes? I’d like to preserve required conversion steps while making optional optimizer selection effective.