fastmachinelearning/hls4ml #1135: KeyError: 'batch_input_shape'

franhervas99, 2024-11-21:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Version 0.8.1 from pip.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.
- [x] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
I have a keras model which has, in the input layer, in config key, a key that is "batch_shape", but when I run config_from_keras_model function, an error is thrown "KeyError: 'batch_input_shape'". Changing the lines 11 and 21 from core.py file, to be batch_shape instead of batch_input_shape, fixes the error.

## Details
Please add to the following sections to describe the bug as accurately as possible.

### Steps to Reproduce
1. Download the model: 
[test_model.zip](https://github.com/user-attachments/files/17846878/test_model.zip)
2. Run typical script to get the config:
```python
keras_model = keras.models.load_model(model_path)
config = hls4ml.utils.config_from_keras_model(keras_model, granularity='name')
```

### Expected behavior
Generate the model config for converting the model to HLS.

### Actual behavior
An error is thrown:
```bash
Traceback (most recent call last):
  File "/home/fran/work_repository/machine_learning_fpga/python/hls4ml_dev.py", line 40, in <module>
    config = hls4ml.utils.config_from_keras_model(keras_model, granularity='name')
  File "/home/fran/.local/lib/python3.10/site-packages/hls4ml/utils/config.py", line 159, in config_from_keras_model
    layer_list, _, _, _ = hls4ml.converters.parse_keras_model(model_arch, reader)
  File "/home/fran/.local/lib/python3.10/site-packages/hls4ml/converters/keras_to_hls.py", line 288, in parse_keras_model
    layer, output_shape = layer_handlers[keras_class](keras_layer, input_names, input_shapes, reader)
  File "/home/fran/.local/lib/python3.10/site-packages/hls4ml/converters/keras/core.py", line 11, in parse_input_layer
    layer['input_shape'] = keras_layer['config']['batch_input_shape'][1:]
KeyError: 'batch_input_shape'
```

## Optional

### Possible fix
- Go to this file: hls4ml/converters/keras/core.py
- Edit the line 11 from ```layer['input_shape'] = keras_layer['config']['batch_input_shape'][1:]``` to ```layer['input_shape'] = keras_layer['config']['batch_shape'][1:]```
- Edit the line 21 from ```output_shape = keras_layer['config']['batch_input_shape']``` to ```output_shape = keras_layer['config']['batch_shape']```

franhervas99, 2024-11-22:
Updating to the last version on main branch gives me the same error.

vloncar, 2024-11-22:
Upon investigation, this is caused by your environment being based on new TF and Keras v3, which are not supported. There's some progress on v3 support in #1116. To install usable hls4ml from the main branch, you will need python 3.10 or 3.11 and TF 2.8 to 2.14. See #1136 for a list of dependencies of the current version.

franhervas99, 2024-11-25:
Hello @vloncar ,
Chaning TF version to 2.12 solves the issue. Thank you so much for your solution!

closed 2024-11-25 by franhervas99