fastmachinelearning/hls4ml #1150: TF model implementation in tf version 2.18.0

ixiotidi, 2024-12-13:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [ ] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [ ] Check that the issue hasn't already been reported, by checking the currently open issues.
- [ ] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
I'm trying to compile a tensorflow 2.18 model that I created using the Function-API with HLS4ML and I"m getting the following error:

    hlsConfig = comp.utils.config_from_keras_model(model, granularity='model')
  File "/workspace/Conda/envs/myenv/lib/python3.9/site-packages/hls4ml/utils/config.py", line 159, in config_from_keras_model
    layer_list, _, _, _ = hls4ml.converters.parse_keras_model(model_arch, reader)
  File "/workspace/Conda/envs/myenv/lib/python3.9/site-packages/hls4ml/converters/keras_to_hls.py", line 260, in parse_keras_model
    input_shapes = [output_shapes[inbound_node[0]] for inbound_node in keras_layer['inbound_nodes'][0]]
IndexError: list index out of range

## Details
When I convert my model into the json format with the to_json() method of tf. I can see that the inbound_node entry in my input_layer is empty and I believe this is causing the issue. If I try to compile my model with Sequential then it's looking for the batch_input_shape() which again doesn't exist in later versions of tf. 

### Steps to Reproduce
Take the CNN example from the HLS4ML tutorial with the update TF version and the error is already there.

1. pip install hls4ml (version 0.8.1)

### Expected behavior
I would have expected the same behaviour as I see with tf version 2.1.0 which is what is coded in the keras_parser()

### Actual behavior
I get the error seeing above

## Optional

### Possible fix
Compare model json format against tf version dependent template in order to extract the correct names rather than using hard-coded dictionary entries?

### Additional context
Add any other context about the problem here.

JanFSchulte, 2024-12-13:
Hi! Thanks for getting in touch. First of all, we just released version 1.0.0 of hls4ml and generally recommend switching to it. In your case, unfortunately this would not help as we currently do not support tensorflow versions above 2.14.1 even in the new release, see https://github.com/fastmachinelearning/hls4ml/blob/main/setup.cfg#L33. 

We are aware of this limitation and will work to improve this for future releases, but at the moment exactly these kind of changes in TF make it difficult to support a wide range of versions.

calad0i, 2024-12-17:
tensorflow 2.18 is based on keras v3. Experimental support is added in PR #1116, but it is not yet stable.

closed 2026-10-15 by github-actions[bot]