fastmachinelearning/hls4ml #1168: 14000 parameter model fails to synthesize even with reuse factor 1000 and strategy resource - ERROR: [XFORM 203-504] Stop unrolling loop 'ResizeHeight'

joekang21, 2025-01-20:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.

- [x] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
Please give a brief and concise description of the bug.

I have a similar problem to https://github.com/fastmachinelearning/hls4ml/issues/1013 but the steps to resolve are not working for me. 

## Details
Please add to the following sections to describe the bug as accurately as possible.

I understand there is a maximum number of parallel multiplications possible (4096). However, the steps to resolve this issue is not working for me. I only have a 14000 parameter UNet network where the max number of parameters per layer is under 5000. I increased usage factor as much as possible and use strategy resource for the entire model. I also tried to reduce the number of parameters so that each layer would have less than 4096. This only worked when I significantly reduced the model so the total number of parameters in the entire model was about 3000. I am confused as to why setting the strategy to resource is not doing anything. My io_type is set to io_stream as per the CNN tutorial.

### Steps to Reproduce
Add what needs to be done to reproduce the bug. Add *commented* code examples and make sure to include the original model files / code, and the commit hash you are working on.

# Run below code on the most updated version of hls4ml - I have removed the training and evaluating for simplicity, let me know if you want to see that as well.

        # BELOW IS THE MODEL ARCHITECTURE
        from keras.models import Model
        from keras.layers import Input, Conv2D, MaxPooling2D, UpSampling2D, concatenate, ReLU
        
        def reduced_unet_model(input_size=(384, 384, 1)):
            inputs = Input(input_size)
            
            # Contracting path
            # Block 1
            c1 = Conv2D(8, (3, 3), padding='same')(inputs)  # Reduced filters to 4 (6.2% of 64)
            c1 = ReLU()(c1)
            p1 = MaxPooling2D((2, 2))(c1)
            
            # Block 2
            c2 = Conv2D(16, (3, 3), padding='same')(p1)  # Reduced filters to 8 (6.2% of 128)
            c2 = ReLU()(c2)
            p2 = MaxPooling2D((2, 2))(c2)
            
            # Bottleneck
            c3 = Conv2D(32, (3, 3), padding='same')(p2)  # Reduced filters to 16 (6.2% of 256)
            c3 = ReLU()(c3)
            
            # Expansive path
            # Block 1
            u1 = UpSampling2D((2, 2))(c3)
            u1 = Conv2D(16, (2, 2), padding='same')(u1)
            u1 = ReLU()(u1)
            merge1 = concatenate([c2, u1], axis=3)
            c4 = Conv2D(16, (3, 3), padding='same')(merge1)
            c4 = ReLU()(c4)
            
            # Block 2
            u2 = UpSampling2D((2, 2))(c4)
            u2 = Conv2D(8, (2, 2), padding='same')(u2)
            u2 = ReLU()(u2)
            merge2 = concatenate([c1, u2], axis=3)
            c5 = Conv2D(8, (3, 3), padding='same')(merge2)
            c5 = ReLU()(c5)
            
            # Output Layer
            outputs = Conv2D(4, (1, 1))(c5)
            outputs = Activation('softmax', name='softmax')(outputs)
            
            model = Model(inputs=[inputs], outputs=[outputs])
            return model
        
        # Instantiate and summarize the model
        model = reduced_unet_model()
        model.summary()
        
        # BELOW IS THE HLS CONFIG 
        
        import hls4ml
        import plotting as plotting #  - plotting file from hls4ml-tutorial
        
        # Then the QKeras model
        hls_config = hls4ml.utils.config_from_keras_model(model, granularity='name')
        
        hls_config['Model']['Strategy'] = 'Resource' 
        hls_config['Model']['ReuseFactor'] = 1000 # For entire model
        
        # for every layer 
        for Layer in hls_config['LayerName'].keys():
            hls_config['LayerName'][Layer]['Trace'] = True
            hls_config['LayerName'][Layer]['Strategy'] = 'Resource'
            hls_config['LayerName'][Layer]['ReuseFactor'] = 1000
            hls_config['LayerName'][Layer]['Precision'] = 'ap_fixed<16,6>'
        
        # stable for softmax
        hls_config['LayerName']['softmax']['Strategy'] = 'Stable'
        
        plotting.print_dict(hls_config)
        
        cfg = hls4ml.converters.create_config(backend='Vivado')
        cfg['IOType'] = 'io_stream'  # Must set this if using CNNs!
        cfg['HLSConfig'] = hls_config
        cfg['KerasModel'] = model
        cfg['OutputDir'] = '/scratch/jxk6137/camus/hls4ml_proj/'
        cfg['XilinxPart'] = 'xcu250-figd2104-2L-e'
        cfg['Part'] = 'xcu250-figd2104-2L-e'
        
        hls_model = hls4ml.converters.keras_to_hls(cfg)
        hls_model.compile()
        
        import os
        
        os.environ['PATH'] = os.environ['XILINX_VIVADO'] + '/bin:' + os.environ['PATH']
        
        hls_model.build(csim=False)

### Expected behavior
Please add a brief description of what you expected to happen.

synthesized model

### Actual behavior
Describe what actually happens instead.

INFO: [HLS 200-489] Unrolling loop 'ImageWidth' (firmware/nnet_utils/nnet_image_stream.h:22) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 32u>, config13>' completely with a factor of 96.
INFO: [HLS 200-489] Unrolling loop 'ImageChan' (firmware/nnet_utils/nnet_image_stream.h:28) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 32u>, config13>' completely with a factor of 32.
INFO: [HLS 200-489] Unrolling loop 'ResizeHeight' (firmware/nnet_utils/nnet_image_stream.h:36) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 32u>, config13>' completely with a factor of 2.
ERROR: [XFORM 203-504] Stop unrolling loop 'ResizeHeight' (firmware/nnet_utils/nnet_image_stream.h:36) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 32u>, config13>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [Common 17-206] Exiting vivado_hls at Mon Jan 20 14:07:38 2025...
CSynthesis report not found.
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.

bo3z, 2025-01-22:
The bottleneck in this case are the `UpSampling` layers, not the `Conv` layers (the function `resize_nearest` is from `UpSampling`). The error here indicates that the body of the loop `ResizeHeight` is too complex so it cannot be fully parallelised (unrolled). Looking at the source code in nnet_image_stream, it contains multiple sub-loops that are also fully unrolled. To tackle this problem, there are several options:
1. You could try using Vitis HLS instead of Vivado HLS, as it may offer better scheduling and resource allocation of the compilation
2. You could simply remove the `#pragma HLS UNROLL` on line 37 of nnet_image_stream.h. While this usually comes at a significant latency penalty, in this case the `ResizeHeight`loop only has 2 iterations, so the overall impact on the model latency might not be as huge (the loop will take twice as long but at the end of the day that layer is only part of the model). If this still doesn't work, you can remove some of the inner #pragma HLS UNROLL lines inside `ResizeHeight` sub-loops. 

Additionally, you may be able to use a smaller Reuse Factor for the Conv layers (since they are not the bottleneck), to get lower latency.

joekang21, 2025-01-22:
Thank you for the quick response. I am currently unable to use Vitis as I am SSHed into a university computer that does not have Vitis installed at the moment. I also removed line 37 and the same error still occurred. I tried disabling unrolling in all loops within nnet_image_stream.h and somehow the same error still occurs. Were you able to get a build to work after removing line 37?

bo3z, 2025-01-24:
Just to confirm: Did you remove the line 37 in the hls4ml templates (https://github.com/fastmachinelearning/hls4ml/blob/main/hls4ml/templates/vivado/nnet_utils/nnet_image_stream.h) and not in the "compiled" model directory (`output_dir/firmware/nnet_utils/nnet_image_stream.h`)? As re-creating and recompiling the hls4ml model will overwrite any changes in the local directory that is specific to your model. If so, can you please post the new error message? Removing the UNROLL pragma could move the error somewhere else (it could be the same error but now for a different loop, i.e. one different to `ResizeHeight`), so line 37 should not be problematic anymore.

joekang21, 2025-01-24:
Previously, I was removing line 37 in my output directory, but this was after compiling my hls model. I now have forked the repository and removed the line in the templates. Below is my updated nnet_image_stream.h file and the error I am getting. It seems to not have changed anything.

        #ifndef NNET_IMAGE_STREAM_H_
        #define NNET_IMAGE_STREAM_H_
        
        #include "hls_stream.h"
        #include "nnet_common.h"
        
        namespace nnet {
        
        template <class data_T, typename CONFIG_T> void resize_nearest(hls::stream<data_T> &image, hls::stream<data_T> &resized) {
            assert(CONFIG_T::new_height % CONFIG_T::height == 0);
            assert(CONFIG_T::new_width % CONFIG_T::width == 0);
            constexpr unsigned ratio_height = CONFIG_T::new_height / CONFIG_T::height;
            constexpr unsigned ratio_width = CONFIG_T::new_width / CONFIG_T::width;
        
        ImageHeight:
            for (unsigned h = 0; h < CONFIG_T::height; h++) {
                #pragma HLS PIPELINE
        
                data_T data_in_row[CONFIG_T::width];
        
            ImageWidth:
                for (unsigned i = 0; i < CONFIG_T::width; i++) {
                    #pragma HLS UNROLL
        
                    data_T in_data = image.read();
        
                ImageChan:
                    for (unsigned j = 0; j < CONFIG_T::n_chan; j++) {
                        #pragma HLS UNROLL
        
                        data_in_row[i][j] = in_data[j];
                    }
                }
        
            ResizeHeight:
                for (unsigned i = 0; i < ratio_height; i++) {
                    
                ImageWidth2:
                    for (unsigned l = 0; l < CONFIG_T::width; l++) {
                        #pragma HLS UNROLL
        
                    ResizeWidth:
                        for (unsigned j = 0; j < ratio_width; j++) {
                            #pragma HLS UNROLL
        
                            data_T out_data;
                            PRAGMA_DATA_PACK(out_data)
        
                        ResizeChan:
                            for (unsigned k = 0; k < CONFIG_T::n_chan; k++) {
                                #pragma HLS UNROLL
        
                                out_data[k] = data_in_row[l][k];
                            }
        
                            resized.write(out_data);
                        }
                    }
                }
            }
        }
        
        } // namespace nnet
        
        #endif
        

ERROR: [XFORM 203-504] Stop unrolling loop 'ResizeHeight' (firmware/nnet_utils/nnet_image_stream.h:36) in function 'nnet::resize_nearest<nnet::array<ap_fixed<16, 6, (ap_q_mode)5, (ap_o_mode)3, 0>, 32u>, config13>' because it may cause large runtime and excessive memory usage due to increase in code size. Please avoid unrolling the loop or form sub-functions for code in the loop body.
ERROR: [HLS 200-70] Pre-synthesis failed.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

joekang21, 2025-01-25:
Update - I was able to install vitis hls on the computer and the issue has been resolved. It is still confusing as to why removing the unroll pragma did not do anything though.

closed 2025-01-25 by joekang21