fastmachinelearning/hls4ml #1194: Error while predicting using VivadoAccelerator backend in hls4ml v1.0.0 tutorial (Part-6) - ERROR: file w2.txt does not exist

alvarorramos, 2025-02-13:
Commit version: e690afe570683581ab3b11784b026aad6b313da7

## Description
Hello, I'm currently following the Part-6 tutorial of the hls4ml v1.0.0 documentation. My goal is to generate a bitstream using the VivadoAccelerator backend. However, I am encountering an issue when trying to predict with the following line of code:
`y_predict_hls4ml = hls_model.predict(np.ascontiguousarray(X_test_reduced))`
The error message returned is:
`ERROR: file w2.txt does not exist.`

Here are some details about my setup:
* hls4ml version: v1.0.0 or 1.1.0.dev20+ge690afe5
* Backend: VivadoAccelerator

### Steps to Reproduce
1. Clone the hls4ml repository
2. Checkout the master branch, with commit hash: e690afe570683581ab3b11784b026aad6b313da7
3. Run conversion [part6_cnns.ipynb ](https://github.com/fastmachinelearning/hls4ml-tutorial/blob/main/part6_cnns.ipynb)
In this tutorial we modified the compilation configuration of the Qkeras model in order to generate the bitstream with VivadoAccelerator backend. This configuration worked with version 0.8.1.

```
# Then the QKeras model
hls_config_q = hls4ml.utils.config_from_keras_model(qmodel, granularity='name')
hls_config_q['Model']['ReuseFactor'] = 1
hls_config_q['Model']['Precision'] = 'ap_fixed<16,6>'
hls_config_q['LayerName']['output_softmax']['Strategy'] = 'Stable'
plotting.print_dict(hls_config_q)

hls_model_q = hls4ml.converters.convert_from_keras_model(
        qmodel, hls_config=hls_config_q, output_dir='quantized_pruned_cnn/', 
        io_type= 'io_stream', backend='VivadoAccelerator', 
        board='pynq-z2'
        )

hls_model_q.compile()
```
This cell is executed correctly but when we try to predict using this hls_model_q, it returns the aforementioned error. 

Would appreciate any help or insights into resolving this issue!
Thanks in advance!

bo3z, 2025-02-17:
Hi @alvarorramos can you check out the PR #1198 and see whether that fixes the mentioned bug?

alvarorramos, 2025-02-19:
Hi @bo3z, I tried checking out to the PR you mentioned and everything seems to work correctly. I am closing the issue, thank you!

closed 2025-02-19 by alvarorramos