fastmachinelearning/hls4ml #1216: Vivado HLS Export IP Fails: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:myproject_axi:1.0

alvarorramos, 2025-03-05:
## Quick summary
I am trying to use hls4ml inside a Docker container with Vivado 2020.1 and hls4ml 0.8.0. While the synthesis step completes successfully, the process fails when exporting the IP with the following error:
`ERROR: [BD 5-390] IP definition not found for VLNV: xilinx.com:hls:myproject_axi:1.0`

## Details
I suspect the issue may be related to how Vivado was installed inside the Docker container, as the generated IP is failing to export. Below is part of the error log:
```
INFO: [VHDL 208-304] Generating VHDL RTL for myproject_axi.
INFO: [VLOG 209-307] Generating Verilog RTL for myproject_axi.
***** C/RTL SYNTHESIS COMPLETED IN 0h8m22s *****
***** EXPORT IP *****
INFO: [IMPL 213-8] Exporting RTL as a Vivado IP.
command 'ap_source' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $arg] "

INFO: [HLS 200-112] Total elapsed time: 504.89 seconds; peak allocated memory: 1.217 GB.
INFO: [Common 17-206] Exiting vivado_hls at Wed Mar  5 14:47:20 2025...
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.
```

### Steps to Reproduce
Reproducing the issue may be cumbersome, but here’s what I did:

1. Installed Vivado 2020.1 inside a Docker container.
2. Created a conda environment with the necessary dependencies.
3. Ran hls4ml tutorial for CNNs using VivadoAccelerator backend
4. The synthesis step completes successfully, but the process fails at the IP export stage.

### Expected Behavior
Vivado should recognize the HLS-generated IP and procees with the export without errors. 

### Questions & Possible Causes
* Could this be related to how Vivado was installed inside Docker?
* Is there a missing dependency or environment variable needed for the IP export?
* Has anyone successfully run hls4ml inside a Dockerized Vivado setup with vivado 2020.1?

Has anyone encountered a similar issue or have any insights on how to fix this? Any help would be greatly appreciated!
Thanks in advance!

vloncar, 2025-03-06:
In the same setup, do the previous steps (which are only related to the IP, not the accelerator backend) work if you add `export=True` to `build()`?

My guess is that the issue is related either to paths in the generated tcl files, or the IP export silently fails due to installation isues (for example, the well known 2020.1 date issue that affects exports for which Xilinx released a patch)

alvarorramos, 2025-03-18:
We have confirmed that the previous steps work correctly when adding export=True to build().

Regarding your mention of the patch, are you referring to the Y2K22 patch? If so, we have already applied it to our installation.

Additionally, do you think the installation paths might differ depending on whether Vivado is installed via the GUI versus using a configuration file?

Thanks in advance for your help!

AdrFebles, 2025-03-20:
We are getting the same error in a Virtual Machine with Ubuntu 18.04, Vivado 2020.1 and HLS4ML 0.8.0. We haven't found a solution yet. Does anyone have experience launching HLS4ML builds on virtual machines?

nghielme, 2025-03-20:
It looks very similar to this: https://adaptivesupport.amd.com/s/question/0D54U00006RvkkWSAR/how-can-i-add-properly-the-patch-y2k22patch12-to-vivado?language=en_US