fastmachinelearning/hls4ml #1231: report not found

howard-liang-B, 2025-03-15:
```
os.environ['PATH'] = os.environ['XILINX_VIVADO'] + '/bin:' + os.environ['PATH']
print("PATH: ", os.environ['PATH'])

float_hls_model.build(csim=False, synth=True, vsynth=True, export=export_ip)
```

```
PATH:  /usr/Xilinx/Vivado/2024.2/bin:/home/user3/stu_ai/.conda/envs/o_hls4ml_tf/bin:/home/user3/stu_ai/.local/bin:/usr/Xilinx/Vitis_HLS/2024.2/bin:/usr/Xilinx/Model_Composer/2024.2/bin:/usr/Xilinx/Vitis/2024.2/bin:/usr/Xilinx/DocNav:/usr/Xilinx/Vivado/2024.2/bin:/usr/Xilinx/PDM/2024.2/bin:/home/user3/stu_ai/.conda/envs/o_hls4ml_tf/bin:/usr/local/anaconda/condabin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin:/home/user3/stu_ai/.dotnet/tools:/home/user3/stu_ai/howard/cuda-12.5/bin

****** Vitis HLS - High-Level Synthesis from C, C++ and OpenCL v2024.2 (64-bit)
  **** SW Build 5238294 on Nov  8 2024
  **** IP Build 5239520 on Sun Nov 10 16:12:51 MST 2024
  **** SharedData Build 5239561 on Fri Nov 08 14:39:27 MST 2024
  **** Start of session at: Sat Mar 15 17:46:54 2025
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.
    ** Copyright 2022-2024 Advanced Micro Devices, Inc. All Rights Reserved.

source /usr/Xilinx/Vitis/2024.2/scripts/vitis_hls/hls.tcl -notrace
INFO: [HLS 200-10] For user 'stu_ai' on host 'chenshihlunlab' (Linux_x86_64 version 6.8.0-51-generic) on Sat Mar 15 17:46:56 CST 2025
INFO: [HLS 200-10] On os Ubuntu 24.04.1 LTS
INFO: [HLS 200-10] In directory '/home/user3/stu_ai/howard/CIFAR-10/HLS/Table_2/float_alexnet'
WARNING: [HLS 200-2053] The vitis_hls executable is deprecated. Consider using vitis-run --mode hls --tcl
Sourcing Tcl script 'build_prj.tcl'
INFO: [HLS 200-1510] Running: open_project myproject_prj 
INFO: [HLS 200-10] Opening project '/home/user3/stu_ai/howard/CIFAR-10/HLS/Table_2/float_alexnet/myproject_prj'.
INFO: [HLS 200-1510] Running: set_top myproject 
INFO: [HLS 200-1510] Running: add_files firmware/myproject.cpp -cflags -std=c++0x 
INFO: [HLS 200-10] Adding design file 'firmware/myproject.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb myproject_test.cpp -cflags -std=c++0x 
INFO: [HLS 200-10] Adding test bench file 'myproject_test.cpp' to the project
INFO: [HLS 200-1510] Running: add_files -tb firmware/weights 
INFO: [HLS 200-10] Adding test bench file 'firmware/weights' to the project
INFO: [HLS 200-1510] Running: add_files -tb tb_data 
INFO: [HLS 200-10] Adding test bench file 'tb_data' to the project
INFO: [HLS 200-1510] Running: open_solution solution1 
INFO: [HLS 200-10] Opening solution '/home/user3/stu_ai/howard/CIFAR-10/HLS/Table_2/float_alexnet/myproject_prj/solution1'.
INFO: [HLS 200-1505] Using flow_target 'vivado'
Resolution: For help on HLS 200-1505 see docs.xilinx.com/access/sources/dita/topic?Doc_Version=2024.2%20English&url=ug1448-hls-guidance&resourceid=200-1505.html
INFO: [HLS 200-1464] Running solution command: config_compile -name_max_length=80
INFO: [XFORM 203-1161] The maximum of name length is set to 80.
INFO: [HLS 200-1510] Running: config_array_partition -maximum_size 4096 
ERROR: [HLS 200-101] config_array_partition: Unknown option '-maximum_size'.
ERROR: [HLS 200-101] config_array_partition: Unknown option '4096'.
[1msyn.array_partition.complete_threshold[0m=[1msyn.array_partition.throughput_driven[0m=
INFO: [HLS 200-1510] Running: config_compile -name_max_length 80 
INFO: [XFORM 203-1161] The maximum of name length is set to 80.
INFO: [HLS 200-1510] Running: set_part XCZU7EV-2FFVC1156 
ERROR: [HLS 200-1023] Part 'XCZU7EV-2FFVC1156' is not installed.
command 'create_platform' returned error code
    while executing
"source build_prj.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel \#0 [list source $tclfile] "

INFO: [HLS 200-112] Total CPU user time: 2.54 seconds. Total CPU system time: 0.15 seconds. Total elapsed time: 2.42 seconds; peak allocated memory: 184.480 MB.
CSynthesis report not found.
Vivado synthesis report not found.
Cosim report not found.
Timing report not found.
```

why is the report not found

bo3z, 2025-03-15:
There are several issues here:
INFO: [HLS 200-1510] Running: config_array_partition -maximum_size 4096, is throwing an exception that this is an unknown command, which should be present in a valid Vivado/Vivado HLS installation.

ERROR: [HLS 200-1023] Part 'XCZU7EV-2FFVC1156' is not installed, indicating that the vivado installation does not include support for this part.

Could you share details what Vivado/Vivado HLS version you are using and verify it's been correctly set-up? Especially regarding the latter issue, as it's not related to hls4ml - either part number is not correct or, the Vivado installation requires some additional components and should be downloaded from the internet.

howard-liang-B, 2025-03-15:
I realized that I spelled the part incorrectly.
thanks

howard-liang-B, 2025-03-16:
Here is my model
```
def get_float_model():
	float_model = keras.models.Sequential([
		Conv2D(filters=48, kernel_size=(3,3), strides=(1,1), activation='relu', input_shape=(32,32,3), 
				kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='conv_1'),
		BatchNormalization(name='bn_1'),
		MaxPooling2D(pool_size=(2,2), name='pool_1'),

		Conv2D(filters=64, kernel_size=(3,3), strides=(1,1), activation='relu', padding="same", 
				kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='conv_2'),
		BatchNormalization(name='bn_2'),
		MaxPooling2D(pool_size=(2,2), name='pool_2'),

		Conv2D(filters=64, kernel_size=(3,3), strides=(1,1), activation='relu', padding="same", 
				kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='conv_3'),
		BatchNormalization(name='bn_3'),

		Conv2D(filters=64, kernel_size=(1,1), strides=(1,1), activation='relu', padding="same", 
				kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='conv_4'),
		BatchNormalization(name='bn_4'),

		Conv2D(filters=64, kernel_size=(1,1), strides=(1,1), activation='relu', padding="same", 
				kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='conv_5'),
		BatchNormalization(name='bn_5'),
		MaxPooling2D(pool_size=(2,2), name='pool_5'),

		Flatten(),
		Dense(128, activation='relu', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='dense_1'),
		Dropout(0.25, name='dropout_1'),

		Dense(128, activation='relu', kernel_initializer='lecun_uniform', kernel_regularizer=l1(0.0001), name='dense_2'),
		Dropout(0.25, name='dropout_2'),

		Dense(10, activation='softmax', name='output_softmax')
	], name='Float_AlexNet')
	return float_model

float_model = get_float_model()
float_model.summary()
```
model summary
```
Model: "Float_AlexNet"
_________________________________________________________________
 Layer (type)                Output Shape              Param #   
=======================================
 conv_1 (Conv2D)             (None, 30, 30, 48)        1344      
                                                                 
 bn_1 (BatchNormalization)   (None, 30, 30, 48)        192       
                                                                 
 pool_1 (MaxPooling2D)       (None, 15, 15, 48)        0         
                                                                 
 conv_2 (Conv2D)             (None, 15, 15, 64)        27712     
                                                                 
 bn_2 (BatchNormalization)   (None, 15, 15, 64)        256       
                                                                 
 pool_2 (MaxPooling2D)       (None, 7, 7, 64)          0         
                                                                 
 conv_3 (Conv2D)             (None, 7, 7, 64)          36928     
                                                                 
 bn_3 (BatchNormalization)   (None, 7, 7, 64)          256       
                                                                 
 conv_4 (Conv2D)             (None, 7, 7, 64)          4160      
                                                                 
 bn_4 (BatchNormalization)   (None, 7, 7, 64)          256       
                                                                 
 conv_5 (Conv2D)             (None, 7, 7, 64)          4160      
                                                                 
 bn_5 (BatchNormalization)   (None, 7, 7, 64)          256       
                                                                 
 pool_5 (MaxPooling2D)       (None, 3, 3, 64)          0         
                                                                 
 flatten (Flatten)           (None, 576)               0         
                                                                 
 dense_1 (Dense)             (None, 128)               73856     
                                                                 
 dropout_1 (Dropout)         (None, 128)               0         
                                                                 
 dense_2 (Dense)             (None, 128)               16512     
                                                                 
 dropout_2 (Dropout)         (None, 128)               0         
                                                                 
 output_softmax (Dense)      (None, 10)                1290      
                                                                 
========================================
Total params: 167,178
Trainable params: 166,570
Non-trainable params: 608
_________________________________________________________________
```

Is it normal for the HLS build to take this long to run with my model architecture?

[image]

bo3z, 2025-03-16:
Yes, this is a large CNN model. You almost certainly have to use io_stream instead of io_parallel, and also tune the Reuse Factor to make sure the model fits onto the hardware. Additionally, where possible, the model should also be quantized during training, e.g. using QKeras. All of these concepts are explained in the hls4ml tutorial: https://github.com/fastmachinelearning/hls4ml-tutorial, which I would recommend to get a feeling for how large the models in hls4ml can be and how to tune these variables to ensure successful synthesis.