fastmachinelearning/hls4ml #1242: High IO utilization even with io_stream

Sayanti17, 2025-03-25:
Description:

    When using 3-bit quantization, the IO utilization reaches 122% with xcu250-figd2104-2L-e FPGA board.
    With 4-bit quantization, the IO utilization increases to 200% with xcu250-figd2104-2L-e FPGA board.
    I attempted using a larger FPGA, but IO utilization remains above 100%.

Current Configurations:
          config['Model']['Precision'] = 'ap_fixed<4,1>'
          config['Model']['ReuseFactor'] = 32  # Optimized reuse factor
          config['Model']['Strategy'] = 'Resource'
          
          # Apply per-layer optimizations
          for layer in autoencoder.model.layers:
              if isinstance(layer, QDense):  # Ensure QDense layers are optimized
                  layer_name = layer.name
                  config['LayerName'][layer_name] = {
                      'ReuseFactor': 16 if 'q_dense_2' not in layer_name else 2,  # Optimize bottleneck
                      'Strategy': 'Resource',
                      'IOType': 'io_stream',  # Ensure each layer is AXI-Stream
                      'pragma': {
                          'unroll': "factor=1",  # Reduce unrolling factor
                          'pipeline': ""  # Enable pipelining
                      }
                  }
          
          # Ensure output directory exists
          output_dir = 'AXI_Vitis_Outputs/117_4b_vitis_stream_10ns'
          os.makedirs(output_dir, exist_ok=True)
          hls_model = hls4ml.converters.convert_from_keras_model(
              model=autoencoder.model,
              hls_config=config,
              output_dir=output_dir,
              io_type='io_stream',
              backend='Vitis',
              clock_period=10,
              part='xcu250-figd2104-2L-e'
          )

Observations & Concerns:

    1. HLS4ML correctly applies streaming and sets the strategy to "Resource", but it appears to ignore some settings, particularly those aimed at limiting loop unrolling. (it's performing full loop unrolling)
   2.  The generated firmware (myproject.cpp) requires manual modifications to reduce IO utilization, which is time-consuming and error-prone.
    3. Modifying VHDL files manually is complex due to the number of files generated.
    
Request for Guidance:
Could you please advise on the following:

    1. Are there any missing configurations or alternative settings that could help reduce IO utilization?
   2.  Is manual modification of myproject.cpp and VHDL files the only way to optimize IO, or is there a way to enforce these changes directly within HLS4ML?
    3. Are there additional optimizations from HLS4ML’s side that can help limit IO usage?

bo3z, 2025-03-25:
What does I/O utilization refer to in this case? FPGA resource (LUTs, FF etc.) usage, or something else? 

What's your model architecture? How large is your model? You can try a smaller model or maybe a larger reuse factor (RF). Also, looking at the script, it seems the RF of dense layers is very low, which could increase resource usage. Recall, dense layers take the shortest to execute (compared to Conv layers, which sequentially, in streams, process the input image), but have a reasonably large resource usage due to the size of the matrix after the Flatten layer. So you can simply increase the RF for Dense layers, which should significantly reduce resources with a (small) penalty in latency

Sayanti17, 2025-03-26:
> What does I/O utilization refer to in this case? FPGA resource (LUTs, FF etc.) usage, or something else?
>
> What's your model architecture? How large is your model? You can try a smaller model or [...]

my model is a stacked autoencoder with 33026 parameters. i will show my model architecture below and also the settings I have used. In my case, I/O utilization refers to the extent of data transfer happening per cycle within the FPGA, particularly between different layers of the neural network during execution. Even though I have explicitly set io_stream to enforce streaming behavior, the logs indicate that data movement is still occurring in a highly parallel fashion, leading to excessive I/O demands. This suggests that some layers—especially Dense layers—might be operating in a fully parallel manner rather than a properly pipelined (streaming) approach. Since dense layers require large matrix multiplications, a low Reuse Factor (RF) may also be contributing to increased I/O usage by forcing more parallel operations. My concern is that HLS4ML might be either ignoring some loop control settings (such as unrolling) or not fully enforcing streaming behavior, resulting in higher-than-expected I/O resource consumption. I am trying to understand if this is expected behavior or if there are additional optimizations that could help enforce proper streaming and reduce I/O utilization. 
Model is given below:   
Encoder:
    QDense Layer:
        Input size: input_dimension
        Activation: ReLU
        Kernel and bias quantization: 4-bit
    QDense Layer:
        Output size: 55% of input_dimension
        Activation: ReLU
        Kernel and bias quantization: 4-bit
    QDense Layer:
        Output size: 27% of input_dimension
        Activation: ReLU
        Kernel and bias quantization: 4-bit

Decoder:
    QDense Layer:
        Output size: 55% of input_dimension
        Activation: ReLU
        Kernel and bias quantization: 4-bit
    QDense Layer:
        Output size: input_dimension
        Activation: Sigmoid (for reconstruction)
        Kernel and bias quantization: 4-bit
also i have tried to increase the reuse factor for the dense layers, but io usage limit exccedds. 
layer_reuse_factors = {
    'q_dense': 64,
    'q_dense_1': 64,
    'q_dense_2': 32,
    'q_dense_3': 32,
    'q_dense_4': 64
}

jmitrevs, 2025-03-28:
We concluded that this is external to hls4ml but part of the external system:

```
4. I/O
------
+------------+------+-------+-----------+--------+
|  Site Type | Used | Fixed | Available |  Util% |
+------------+------+-------+-----------+--------+
| Bonded IOB | 1060 |     0 |       676 | 156.80 |
+------------+------+-------+-----------+--------+
```
therefore, closing this issue.

closed 2025-03-28 by jmitrevs