fastmachinelearning/hls4ml #1244: TLAST Signal Missing in IP Core Generated by HLS4ML

SIRIDHARI, 2025-03-25:
Hi,
I am using Vitis HLS for my signature forgery detection project. I wrote the ML model in Python and used HLS4ML to convert it into C++ files. I then used Vitis HLS to generate RTL files.

I am planning to include DMA in my block design, so I need the TLAST signal. However, although I am using io_stream in HLS4ML (which should automatically generate the TLAST signal), it is not being generated in the RTL files.

Additionally, I am unsure where to connect the ap_ctrl signals generated in the IP core within the Vivado block design.

Has anyone faced a similar issue or knows how to ensure TLAST is included in the RTL design? Any help or suggestions would be greatly appreciated.

vusalnovruzov, 2026-01-20:
Hi — did you ever find a solution for getting TLAST generated with io_stream in hls4ml / Vitis HLS?
I’m seeing the same behavior (AXIS data/valid/ready generated, but no TLAST), and I’m also unsure how you ended up wiring the ap_ctrl signals in Vivado.
If you solved it, could you share what change/workaround worked for you?

bo3z, 2026-01-20:
It depends on the HLS version - I've noticed that some (newer) versions of Vitis HLS generate the TLAST. Other than that, I found no reliable way to make sure the HLS compiler generates a TLAST signal.

closed 2026-10-15 by github-actions[bot]