fastmachinelearning/hls4ml #1271: Packaging files setup.cfg  and setup.py have been removed

GiuseppeDiGuglielmo, 2025-04-12:
# Summary

The files `setup.cfg`  and `setup.py` (in the top level directory) have been removed in this commit 9857da11e0d47775a927a70f29a6a5f557f01776

They were useful when running hls4ml with Catapult. I could have local working copies of the repository and point to them with [PYTHONPATH](https://docs.python.org/3/using/cmdline.html#envvar-PYTHONPATH).

## Reproduce

```
git clone --recursive git@github.com:fastmachinelearning/hls4ml.git
cd hls4ml
git checkout 9857da1
PYTHONPATH=$PWD python -c "import hls4ml"
```
It will fail with `importlib.metadata.PackageNotFoundError: No package metadata was found for hls4ml`

At the previous commit (1ad1ad9801130774b61c0a632d8ee4f22921157d), the setup files are still there and I can run:
```
PYTHONPATH=$PWD python -c "import hls4ml"
```

Can we bring them back? Is there an alternative solution so I can keep using PYTHONPATH to point to local hls4ml copies?

Thanks

vloncar, 2025-04-12:
The [official suggestion](https://packaging.python.org/en/latest/guides/writing-pyproject-toml/) for packaging is to use the `pyproject.toml` file to set up the project and the dependencies so we switched to that. The old way is gone and not coming back. If your goal is to install any given repo/branch and make it visible to your scripts (so `import hls4ml` works) you can do it like:
```
git clone --recursive git@github.com:fastmachinelearning/hls4ml.git
cd hls4ml
pip install -e .
git checkout whichever_branch_or_commit_you_want
```
after you do `pip install -e .` (don't forget the dot) your environment will "link" to the installation folder, which is the git repo you cloned. Then all changes within that repo will be reflected in the install, so you can modify files, switch branches etc.

GiuseppeDiGuglielmo, 2025-04-12:
This breaks a little how we were doing things with Catapult. Let me highlight this for @dgburnette and see if they offer an alternative. 

`pip install -e.` is fine but I personally liked `PYTHONPATH` better.

GiuseppeDiGuglielmo, 2025-04-13:
@vloncar can you please confirm that the following commands work as expected in a newly create environment? for example you can use the following instructions) 

```
# Create a throwaway Python environment  
wget https://raw.githubusercontent.com/GiuseppeDiGuglielmo/catapult_ai_nn_setup/refs/heads/main/2024.2_2/create_env.sh
wget https://raw.githubusercontent.com/GiuseppeDiGuglielmo/catapult_ai_nn_setup/refs/heads/main/2024.2_2/requirements.txt
chmod 755 create_env.sh
./create_env.sh python3 $HOME/ccs_venv
source ~/ccs_venv/bin/activate

# Install hls4ml from a local copy of the repo
git clone --recursive git@github.com:fastmachinelearning/hls4ml.git
cd hls4ml
pip install -e .

# Run a Python script that import hls4ml
cd /tmp
python -c "import hls4ml; print(hls4ml.__version__)"
```

The `hls4ml` package doesn't seem properly install for me
```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
AttributeError: module 'hls4ml' has no attribute '__version__'
````

This is not the first time that I use `pip install -e .`, but it is the first time that it fails this way.

Thank you

dgburnette, 2025-07-11:
The problem with using 'pip install -e' is that it requires the user to link their python environment to a specific installation of Catapult. 
This limits the ability of a user to work with multiple different Catapult releases using the same python env.
It also prevents customers from using a centrally installed python env shared by all Catapult users.

dgburnette, 2025-12-12:
@vloncar You are correct that the python code contained within the Catapult install tree is stripped down. But it is also only the one Python package 'hls4ml'. We expect the user to create a Python Virtual Environment with the rest of the required Python packages (we provide a requirements.txt for the correct set). And then we use PYTHONPATH to direct Python to look in the Catapult install tree for 'hls4ml' while the rest of the packages are found via the Virtual Environment. This works fine for Catapult users (because we still ship the setup files). But in Giuseppe's case he wants to selectively pick up hls4ml from within Catapult or from a git workspace - and using PYTHONPATH used to work for that. Now he will have to use 'pip install -e' which does work but means he needs a virtual environment for each Catapult install. An since we only ship the one backend with our tree, he won't be able to switch backends with that virtual environment. I will look to see if the rest of the 'hls4ml' files have any licensing terms that would preclude us from including all of 'hls4ml' in our tree. @GiuseppeDiGuglielmo

GiuseppeDiGuglielmo, 2025-12-12:
Hi guys. For a certain amount of time, it has been easy for me to switch between Catapult distributed code and my git work directories of `hls4ml`. Since `PYTHONPATH` support was removed, this became more cumbersome.

Using _symlinks_ seems the simplest thing we can do. I will give it a try to see if that really work for how I have been doing things. @dgburnette let me know when you have a chance to try it as well. If it works, we adopt that approach, and we close this comment. Thank you

closed 2026-10-14 by github-actions[bot]