fastmachinelearning/hls4ml #1278: AttributeError: module 'hls4ml.model' has no attribute 'profiling'

GiuseppeDiGuglielmo, 2025-04-24:
Escalating this [issue](https://github.com/fastmachinelearning/hls4ml-tutorial/issues/102) noticed when running the tutorial.

jmitrevs, 2025-04-24:
I confirmed this:

```
(fastml311) mac-137349:~ jmitrevs$ ipython
Python 3.11.9 | packaged by conda-forge | (main, Apr 19 2024, 18:45:13) [Clang 16.0.6 ]
Type 'copyright', 'credits' or 'license' for more information
IPython 8.27.0 -- An enhanced Interactive Python. Type '?' for help.

In [1]: import hls4ml

In [2]: hls4ml.model.profiling.get_ymodel_keras()
---------------------------------------------------------------------------
AttributeError                            Traceback (most recent call last)
Cell In[2], line 1
----> 1 hls4ml.model.profiling.get_ymodel_keras()

AttributeError: module 'hls4ml.model' has no attribute 'profiling'
``` 

though I do notice that `from hls4ml.model.profiling import get_ymodel_keras` seems to work.

jmitrevs, 2025-04-24:
You can also `import hls4m.model.profiling` first, and then it works.

jmitrevs, 2025-04-24:
So profiling is not imported by model and needs to be explicitly imported. This is the intended way for it to work. Either uses `from hls4ml.model.profiling import get_ymodel_keras` or `import hls4m.model.profiling` first.