fastmachinelearning/hls4ml #1283: Vivado_hls 2019.2 synthesis error

Smitashree-code, 2025-04-29:
ERROR: [HLS 200-70] Compilation errors found: In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:4:
In file included from firmware/parameters.h:7:
In file included from firmware/nnet_utils/nnet_code_gen.h:4:
In file included from firmware/nnet_utils/nnet_conv1d_latency.h:4:
In file included from firmware/nnet_utils/nnet_common.h:5:
firmware/nnet_utils/nnet_helpers.h:275:10: warning: 'auto' type specifier is a C++11 extension [-Wc++11-extensions]
    for (auto i = 0; i < SIZE; i++)
         ^
firmware/nnet_utils/nnet_helpers.h:374:1: error: unknown type name 'constexpr'
constexpr int ceillog2(int x) { return (x <= 2) ? 1 : 1 + ceillog2((x + 1) / 2); }
^
firmware/nnet_utils/nnet_helpers.h:374:11: error: expected unqualified-id
constexpr int ceillog2(int x) { return (x <= 2) ? 1 : 1 + ceillog2((x + 1) / 2); }
          ^
firmware/nnet_utils/nnet_helpers.h:376:1: error: unknown type name 'constexpr'
constexpr int floorlog2(int x) { return (x < 2) ? 0 : 1 + floorlog2(x / 2); }
^
firmware/nnet_utils/nnet_helpers.h:376:11: error: expected unqualified-id
constexpr int floorlog2(int x) { return (x < 2) ? 0 : 1 + floorlog2(x / 2); }
          ^
firmware/nnet_utils/nnet_helpers.h:378:1: error: unknown type name 'constexpr'
constexpr int pow2(int x) { return x == 0 ? 1 : 2 * pow2(x - 1); }
^
firmware/nnet_utils/nnet_helpers.h:378:11: error: expected unqualified-id
constexpr int pow2(int x) { return x == 0 ? 1 : 2 * pow2(x - 1); }
          ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:4:
In file included from firmware/parameters.h:7:
In file included from firmware/nnet_utils/nnet_code_gen.h:4:
In file included from firmware/nnet_utils/nnet_conv1d_latency.h:4:
firmware/nnet_utils/nnet_common.h:38:12: error: unknown type name 'constexpr'
    static constexpr int leftN = pow2(floorlog2(N - 1)) > 0 ? pow2(floorlog2(N - 1)) : 0;
           ^
firmware/nnet_utils/nnet_common.h:38:22: error: expected unqualified-id
    static constexpr int leftN = pow2(floorlog2(N - 1)) > 0 ? pow2(floorlog2(N - 1)) : 0;
                     ^
firmware/nnet_utils/nnet_common.h:39:12: error: unknown type name 'constexpr'
    static constexpr int rightN = N - leftN > 0 ? N - leftN : 0;
           ^
firmware/nnet_utils/nnet_common.h:39:22: error: expected unqualified-id
    static constexpr int rightN = N - leftN > 0 ? N - leftN : 0;
                     ^
firmware/nnet_utils/nnet_common.h:46:25: error: use of undeclared identifier 'leftN'
    return op(reduce<T, leftN, Op>(x, op), reduce<T, rightN, Op>(x + leftN, op));
                        ^
In file included from firmware/myproject.cpp:1:
In file included from firmware/myproject.cpp:4:
In file included from firmware/parameters.h:7:
In file included from firmware/nnet_utils/nnet_code_gen.h:4:
In file included from firmware/nnet_utils/nnet_conv1d_latency.h:5:
firmware/nnet_utils/nnet_mult.h:32:12: warning: 'auto' type specifier is a C++11 extension [-Wc++11-extensions]
    static auto product(x_T a, w_T w) -> decltype(-a) {
           ^
firmware/nnet_utils/nnet_mult.h:32:12: error: 'auto' not allowed in function return type
firmware/nnet_utils/nnet_mult.h:32:38: error: expected ';' at end of declaration list
    static auto product(x_T a, w_T w) -> decltype(-a) {
                                     ^
                                     ;
firmware/nnet_utils/nnet_mult.h:44:12: warning: 'auto' type specifier is a C++11 extension [-Wc++11-extensions]
    static auto product(x_T a, w_T w) -> decltype(-w) {
           ^
firmware/nnet_utils/nnet_mult.h:44:12: error: 'auto' not allowed in function return type
firmware/nnet_utils/nnet_mult.h:44:38: error: expected ';' at end of declaration list
    static auto product(x_T a, w_T w) -> decltype(-w) {
                                     ^
                                     ;
firmware/nnet_utils/nnet_mult.h:56:12: warning: 'auto' type specifier is a C++11 extension [-Wc++11-extensions]
    static auto product(x_T a, w_T w) -> decltype(-a) {
           ^
firmware/nnet_utils/nnet_mult.h:56:12: error: 'auto' not allowed in function return type
firmware/nnet_utils/nnet_mult.h:56:38: error: expected ';' at end of declaration list
    static auto product(x_T a, w_T w) -> decltype(-a) {
                                     ^
                                     ;
firmware/nnet_utils/nnet_mult.h:70:12: warning: 'auto' type specifier is a C++11 extension [-Wc++11-extensions]
    static auto product(x_T a, w_T w) -> decltype(a * w) {
           ^
firmware/nnet_utils/nnet_mult.h:70:12: error: 'auto' not allowed in function return type
firmware/nnet_utils/nnet_mult.h:70:38: error: expected ';' at end of declaration list
    static auto product(x_T a, w_T w) -> decltype(a * w) {
                                     ^
                                     ;
firmware/nnet_utils/nnet_mult.h:79:17: warning: alias declarations are a C++11 extension [-Wc++11-extensions]
    using r_T = ap_fixed<2 * (decltype(w_T::weight)::width + x_T::width), (decltype(w_T::weight)::width + x_T::width)>;
                ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]
6 warnings and 20 errors generated.
Failed during preprocessing.
    while executing
"source /home/smita/Desktop/model_outputs/hls4ml_project/p2/solution2/csynth.tcl"
    invoked from within
"hls::main /home/smita/Desktop/model_outputs/hls4ml_project/p2/solution2/csynth.tcl"
    ("uplevel" body line 1)
    invoked from within
"uplevel 1 hls::main {*}$args"
    (procedure "hls_proc" line 5)
    invoked from within
"hls_proc $argv"  p2:solution2  Apr 29, 2025, 7:51:56 AM

vloncar, 2025-04-29:
Can you try with 2020.1, or switch to Vitis HLS 2023.2 or newer? Not sure but maybe the 2019.2 doesn't work anymore. I see that the documentation says we support that version but it has been a long time since anyone tried to actually use it.

Smitashree-code, 2025-04-29:
can I try with vitis HLS 2024.2 in window operating system?I have downloaded 2019.2 by seeing the documentation.

JanFSchulte, 2025-04-29:
We support Vitis HLS 2024.1. 2024.2 will probably also work, but it is not fully validated. We do not support using hls4ml on Windows at all, see https://fastmachinelearning.org/hls4ml/intro/status.html.

vloncar, 2025-04-29:
None of the developers use Windows, so we don't know if it works now or not. We know that the the python integration (calling `compile()`) won't work. Installing Vitis in the WSL environment and using it through that does work, that's the equivalent of having a linux installation.