fastmachinelearning/hls4ml #1328: PReLU parameters not generated

franhervas99, 2025-07-14:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.
- [x] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
I'm trying to implement a CNN which implements a PReLU activation function. The parameters of the PReLU are not being propagated correctly to the HLS files.

## Details
### Steps to Reproduce

1. This is the sample code to reproduce the issue:
```python
import hls4ml
import keras
import numpy as np

keras_model = keras.Sequential([
    keras.layers.Input(shape=(9, 1)),
    keras.layers.Conv1D(filters=6, kernel_size=3, strides=1, padding='same'),
    keras.layers.PReLU(shared_axes=[1]),
])

keras_model.summary()

for layer in keras_model.layers:
    weights = layer.get_weights()
    if weights:
        random_weights = [np.random.randn(*w.shape) for w in weights]
        layer.set_weights(random_weights)

for i, layer in enumerate(keras_model.layers):
    weights = layer.get_weights()
    if weights:
        print(f"Layer {i} ({layer.name}) weights:")
        for w in weights:
            print(w)

hls4ml_config = hls4ml.utils.config_from_keras_model(keras_model, granularity='name', default_reuse_factor=1, default_precision='fixed<16,6>')
hls4ml_model = hls4ml.converters.convert_from_keras_model(keras_model, hls_config=hls4ml_config, output_dir='out/', backend='Vitis', part='xcku115-flva1517-2-e', clock_period=2.5, project_name='test_project')

hls4ml.utils.plot_model(hls4ml_model, show_shapes=True, show_precision=True, to_file='hls4ml_model.png')

hls4ml_model.write()
hls4ml_model.compile()
hls4ml_model.build()
```
2. Run it using ```python3 main.py```
3. See the generated .h file for the weights (parameters) of the PReLU. It should look like this. The file is **./out/firmware/weights/a4.h**.
```c
//Numpy array shape [1, 6]
//Min -2.232540130615
//Max 1.219962954521
//Number of zeros 0

#ifndef A4_H_
#define A4_H_

#ifndef __SYNTHESIS__
p_re_lu_param_t a4[6];
#else
p_re_lu_param_t a4[6] = {, , , , , };

#endif

#endif
```
4. Also, the terminal log looks like this:
```bash
ERROR: [HLS 207-1186] expected expression (firmware/weights/a4.h:12:26)
ERROR: [HLS 207-1186] expected expression (firmware/weights/a4.h:12:28)
ERROR: [HLS 207-1186] expected expression (firmware/weights/a4.h:12:30)
ERROR: [HLS 207-1186] expected expression (firmware/weights/a4.h:12:32)
ERROR: [HLS 207-1186] expected expression (firmware/weights/a4.h:12:34)
```
### Expected behavior
It should propagate the parameters from the model to the header files for using them in HLS.

### Actual behavior
It doesn't propagate the parameters.

JanFSchulte, 2025-07-14:
Thanks for bringing this to our attention. There was an issue with PReLU being left out when handling the precision of the parameters, which is fixed by https://github.com/fastmachinelearning/hls4ml/pull/1329.

vloncar, 2025-07-15:
Validation step (`build(..., validation=True)`) failed. Check the `tb_data/csim_results.log` and `tb_data/rtl_cosim_results.log` for differences.

franhervas99, 2025-07-17:
I've detected that the validation step fails when there is a connection between a Conv1D layer and a PReLU activation.I've tried some other combinations  and it fails every time.

franhervas99, 2025-07-17:
Maybe the PReLU parameters are not correctly multiplied in the firmware. The reason is that when shared_axes=[1] (all the sequence in the each channel is multiplied by the same parameter), the validation fails. But, when shared_axes=None (every position in each channel is multiplied by a different parameter), even if every parameter in each channel remains the same, the validation step success. However, the result is not as expected.

closed 2025-07-18 by vloncar