fastmachinelearning/hls4ml #1332: Build options not being propagated when using Vitis backend

franhervas99, 2025-07-16:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.
- [x] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
When applying build() to the hls4ml model, options are not being taken into account. For example, changing cosim=False or validation=False, does not make any changes to the build.

## Details
### Steps to Reproduce
1. Run the next script:
```python
import hls4ml
import keras
import numpy as np
import os
import shutil

if os.path.isdir('out'):
    shutil.rmtree('out')

keras_model = keras.Sequential([
    keras.layers.Input(shape=(9, 1)),
    keras.layers.Conv1D(filters=6, kernel_size=3, strides=1, padding='same'),
    keras.layers.PReLU(shared_axes=[1])
])

keras_model.summary()

for layer in keras_model.layers:
    weights = layer.get_weights()
    if weights:
        random_weights = [np.random.uniform(-1, 1, size=w.shape) for w in weights]
        layer.set_weights(random_weights)

for i, layer in enumerate(keras_model.layers):
    weights = layer.get_weights()
    if weights:
        print(f"Layer {i} ({layer.name}) weights:")
        for w in weights:
            print(w)

hls4ml_config = hls4ml.utils.config_from_keras_model(keras_model, granularity='name', default_reuse_factor=1, default_precision='fixed<32,16>')
hls4ml_model = hls4ml.converters.convert_from_keras_model(keras_model, hls_config=hls4ml_config, output_dir='out/', backend='Vitis', part='xcku115-flva1517-2-e', clock_period=2.5, project_name='test_project')

hls4ml.utils.plot_model(hls4ml_model, show_shapes=True, show_precision=True, to_file='hls4ml_model.png')

hls4ml_model.write()
hls4ml_model.compile()

# in_data = np.random.uniform(-1, 1, size=(9, 1))
# out_data = hls4ml_model.predict(in_data)
# print(out_data)

hls4ml_model.build(cosim=False, validation=False)
```
2. Check the log and see that COSIM and Validation steps are performed.

### Expected behavior
COSIM and the Validation between C/COSIM steps shouldn't be applied.

### Actual behavior
COSIM and Validation steps are applied.

## Optional
### Possible fix
Taking a look at the autogenerated HLS project, at the tcl scripts, I've realized that build_prj.tcl doesn't source build_opt.tcl, which is the file where the options are changed from build() arguments. That is, as I looked, because the VitisWriter class doesn't include the Vitis template for the build_prj.tcl, but the Vivado one instead. That could be because the VivadoWriter writes build_prj.tcl into the HLS project, and the VitisWriter doesn't do it, so takes the Vivado one by default.

franhervas99, 2025-07-24:
I didn't tested it, but it is the fix that I made in my computer and it worked. So it should fix that bug.

bo3z, 2025-07-26:
Fixed in #1342

closed 2025-07-26 by bo3z