fastmachinelearning/hls4ml #1374: cannot open shared object file: No such file or directory

AdrFebles, 2025-09-02:
## Prerequisites
Please make sure to check off these prerequisites before submitting a bug report.
- [x] Test that the bug appears on the current version of the master branch. Make sure to include the commit hash of the commit you checked out.
- [x] Check that the issue hasn't already been reported, by checking the currently open issues.
- [x] If there are steps to reproduce the problem, make sure to write them down below.
- [ ] If relevant, please include the hls4ml project files, which were created directly before and/or after the bug.

## Quick summary
Pytorch + VivadoAccelerator fails to compile

## Details
Trying to compile a Pytorch model with 4 CNN layers and one Dense layer fails with VivadoAccelerator, 'io_parallel' and stratergy Resource for layers with more than 4096 parameters.

### Steps to Reproduce

1. Clone the hls4ml repository
2. Checkout the master branch, with commit hash: [[8a4f2685]](https://github.com/fastmachinelearning/hls4ml/commit/8a4f268567e89ca96aa6c1d96bc3b754a39c8cb6)
3. I can't share the model architecture, but this is the HLS4ML configuration I'm using to convert it.
The size of each layer is:

Layer Name                        # Parameters
==================================================
conv1.weight                                     48
conv1.bias                                          16
conv2.weight                                    1536
conv2.bias                                           32
conv3.weight                                     6144
conv3.bias                                            64
conv4.weight                                    24576
conv4.bias                                            128
dense1_finetuned.weight                 523776
dense1_finetuned.bias                        2

`
config = hls4ml.utils.config_from_pytorch_model(model, input_shape=[(1, 1, 16384)], granularity='name', backend='VivadoAccelerator', transpose_outputs = True)

config['Model']['Precision'] = 'ap_fixed<16,6>'
config['Model']['ReuseFactor'] = 1    
for Layer in config['LayerName'].keys():
        config['LayerName'][Layer]['Strategy'] = 'Latency'
        config['LayerName'][Layer]['ReuseFactor'] = 6000
for layer_name, param in model.named_parameters():
        if param.requires_grad:
            num_params = param.numel()
            if num_params > 4096: 
                name = layer_name.split('.')[0]
                print("Layer {} is too large ({}), are you sure you want to train?".format(name, num_params)) 
                config['LayerName'][name]['Strategy'] = 'Resource'
                config['LayerName'][name]['ReuseFactor'] = 7936

board = 'zcu102'
backend = 'VivadoAccelerator'
OutputDir = ' my-hls-test-quantized-torch'
io_type = 'io_pararell'

hls_model = hls4ml.converters.convert_from_pytorch_model(
        model, hls_config=config, 
        output_dir=f'{OutputDir}/model_v1', 
        backend=backend, 
        board=board
        )
    
hls_model.compile()
`

### Expected behavior
No errors in the model compilation.

### Actual behavior

Error:
`
-> hls_model.compile()
(Pdb) WARNING: You set a Part that does not correspond to the Board you specified. The correct Part is now set.
OSError: my-hls-test-quantized-torch/model_v1/firmware/myproject-870a07f4.so: cannot open shared object file: No such file or directory
`

### Additional context
My PC has a 16GB RAM and I added 32GB of SWAP.
I'm not sure if this is a memory problem.

Thanks in advance for any idea or recommendation.

AdrFebles, 2025-09-03:
Hi @vloncar, the console only displays this error message:

`File "/home/miniconda3/envs/hls4ml-tutorial-main/lib/python3.10/site-packages/hls4ml/model/graph.py", line 801, in compile
    self._compile()
  File "/home/miniconda3/envs/hls4ml-tutorial-main/lib/python3.10/site-packages/hls4ml/model/graph.py", line 820, in _compile
    self._top_function_lib = ctypes.cdll.LoadLibrary(lib_name)
  File "/home/miniconda3/envs/hls4ml-tutorial-main/lib/python3.10/ctypes/__init__.py", line 452, in LoadLibrary
    return self._dlltype(name)
  File "/home/miniconda3/envs/hls4ml-tutorial-main/lib/python3.10/ctypes/__init__.py", line 374, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: my-hls-test-quantized-torch/natwork_model_v1/firmware/myproject-d420b461.so: cannot open shared object file: No such file or directory
`
Regarding RAM usage, I have checked it and it is indeed using almost all available memory. I am quite sure that the compilation fails due to lack of memory. I will try on a computer with more RAM.

[image]

calad0i, 2025-09-08:
There should be trace backs before this message regarding what went wrong during compilation. Though, this one looks like OOM killed.

bo3z, 2026-02-02:
How big is your model? What kind of architecture (CNN, MLP, RNN etc.)?

OOM erro is typically due to the HLS compiler getting "stuck" in converting the HLS to Verilog. If the model is simply too complex, it may be worth changing the io_type, reuse factor or even reducing the model.